DalFSAE/high-voltage-controller #7: Microcontroller Brownout When Switching Contactors

AshtonDudley, 2025-03-25:
# Issue
When the MPU switches the contactors, the eFuse IC causes the MPU to brown out

## Possible Fix

## Images

### **Brownout behaviour** 
(Purple = 3V, Green = 5V, Yellow = AIR_GND, Cyan = SDC_IN)  
[image]

[image]

### **Power Supply Startup**
(Yellow = 12V input, Purple = eFuse output, Cyan = 5V Output, Green = 3V3 Output)
_0.2A PSU limit_
[image]

_3A PSU limit_
[image]

Zoomed in startup_
[image]

###**Power rails during brownout**
(Yellow = 12V input, Purple = eFuse output, Cyan = 5V Output, Green = 3V3 Output)

[image]

[image]

AshtonDudley, 2025-03-26:
@DalFSAE/electrical, does anyone want to take a look at this?

DaniyalV1, 2025-04-02:
Is this issue solved yet? If not, I have a couple of questions:

- I kind of forgot what we talked about last time. Was the issue because eFuse was at a wrong voltage value?
- If the voltage value for the eFuse is wrong, then how much is it supposed to be?

[image]
So, do we have to change some resistor values to change the voltage for the fuse?

AshtonDudley, 2025-04-02:
Hey Daniel, it has not been solved yet. I believe the issue is r1 r2 and r3 on the image you shared. I would need someone to use the schematic to figure out what the current threshold is set to, and then to determine what values are needed to set the under voltage limit to only 8V, instead of the 15V it I currently set to

DaniyalV1, 2025-04-03:
Update for the future: The undervoltage was already set to 8v; the issue is related to something else.

Potential problems: 
- The over-voltage protection is set to 18v; the voltage spike might be causing that. We need to check with an oscilloscope to verify this.
- The capacitor needed to set the startup time is not installed. This makes the startup time for eFuse 1ms, but the recommended time is 39ms. Installing a  0.255µF capacitor sets the startup time to 39ms.

AshtonDudley, 2025-04-03:
I have a spare board we can use for testing.

Things to solve:
- Why is the under-voltage setting not correct?
- Why is the chip causing a fault? 

Things to try:
- remove IMON and check current limit
- vary the resistors 
-add a capactor to dv/dt

> IC Internal Diagram
[image]

DaniyalV1, 2025-04-03:
Another uptade for future:

1. Tested the voltage fluctuations from the contactors and found that the voltage was exceeding 18v, triggering the fuse's overvoltage protection.
2. Inspected the circuit board and discovered a misplaced capacitor where a resistor should have been. Replaced it with a 510KΩ resistor (R1 in the schematic) and adjusted R2 to 82KΩ. This way, the undervoltage protection is 6.62v, and the overvoltage protection is 24.67v.

AshtonDudley, 2025-04-03:
Thanks for the help! Brownout issue has been solved, closing issue.