DalFSAE/high-voltage-controller #10: Test voltage tap cell fuses

AshtonDudley, 2025-03-26:
Create a simple procedure for testing the voltage tap PCBs. The goal is to simulate what would happen if one of our 18650 P26B Li-ion cells was shorted.

We will need:

- Equipment list 
- Test voltages and currents

See #5 for an example of a procedure

AshtonDudley, 2025-03-26:
Let me know if you have any questions!

[image]

[image]

[image]

ElizabetZhekova, 2025-04-03:
Here's the procedure that I came up with today:
Testing procedure for 18650 P26B Li-ion cell shortage

Equipment

Voltage tap PCB
4.2 V Digital load
Closed loop with power supply
Oscilloscope

Procedure

1. Connect the oscilloscope to the tabs of the voltage tap PCB
2. Supply a digital load at 4.2V
3. Increase the amperage linearly until the fuse burns out
4. Record the current supplied before the fuses burn out

I'll come by at 4:15-4:30 tomorrow to test it out.

AshtonDudley, 2025-04-03:
Thanks. This looks good, though we probably want to step up the current we test, as shorts are ordinarily short, with high current events. Fuses are typically tested at a constant current, and the time to opening is measured.

I'd like to test the following: 

- 500mA cont. This is to simulate the BMS balancing, and should not blow the fuse. 
- 1A: Should blow the polyfuse after a few seconds, but not the mechanical fuse.
- 10A: Should blow the fuse instantly
- Short an 18650 P26A. I want to find the max interupt current of the cell
- Short a 21700 P45B. Same reason.

Using the scope we must record:
- Circuit current draw 
- Voltage before the fuse
- Voltage after the fuse

ElizabetZhekova, 2025-04-05:
Heres the data from the test we did today: 
Multimeter in mV

Notes pre test:
-Added jumper between cell 1 and cell 2
-1.63 Ohm resistance between cell 0 and cell 1
-Cell 0 is ground
-Resistance of cell 0 and 1 was too high so we had to switch to 16 gauge wire

[image]

Unregulated power supply test

[image]
With unregulated power supply max current went up to 36A
At 12V it went up to 76 A
At 15V it went up to 100 A
*but main fuses don't work

AshtonDudley, 2025-05-03:
Forgot to close

closed 2025-05-03 by AshtonDudley